trikset/trikRuntime #178: Unable to build on Ubuntu 16.04

NataliaZabelina, 2016-07-06:
Hello,

I am unable to build trikRuntime on my Ubuntu 16.04 LTS (currently latest available version) due to
```
make[1]: Entering directory '/home/natasha/trikRuntime/trikControl'
g++ -c -include .build/x86-release/.obj/trikControl-x86 -m64 -pipe -std=c++1y -Wextra -Wcast-qual -Wwrite-strings -Wredundant-decls -Wunreachable-code -Wnon-virtual-dtor -Woverloaded-virtual -fno-omit-frame-pointer -O2 -Wall -W -D_REENTRANT -fPIC -DTRIKCONTROL_LIBRARY -DQT_NO_DEBUG -DQT_MULTIMEDIA_LIB -DQT_XML_LIB -DQT_GUI_LIB -DQT_CORE_LIB -DQT_SHARED -I/usr/share/qt4/mkspecs/linux-g++-64 -I. -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtGui -I/usr/include/qt4/QtXml -I/usr/include/qt4/QtMultimedia -I/usr/include/qt4 -I. -Iinclude/trikControl -I../qslog -I../trikKernel/include -I../trikHal/include -I.build/x86-release/.moc -o .build/x86-release/.obj/brick.o src/brick.cpp
In file included from src/tonePlayer.h:17:0,
                 from src/brick.cpp:47:
src/audioSynthDevices.h:19:37: fatal error: QtMultimedia/QAudioFormat: No such file or directory
compilation terminated.
Makefile:729: recipe for target '.build/x86-release/.obj/brick.o' failed
make[1]: *** [.build/x86-release/.obj/brick.o] Error 1
make[1]: Leaving directory '/home/natasha/trikRuntime/trikControl'
Makefile:166: recipe for target 'sub-trikControl-make_default' failed
make: *** [sub-trikControl-make_default] Error 2
```

I tried adding package `qtmultimedia5-dev` which is obviously not the one I need (compilation uses qt4, right?). Some say that issue can be resolved by adding `qtmobility-dev` package, but [it was removed for some reason](https://launchpad.net/ubuntu/xenial/amd64/qtmobility-dev/1.2.0-3ubuntu5)

What should I do?

dvvrd, 2016-07-06:
Consider building branch [3.1-dev](https://github.com/trikset/trikRuntime/tree/3.1-dev)

NataliaZabelina, 2016-07-06:
Now it says
```
/usr/bin/ld: cannot find -lQtMultimedia
collect2: error: ld returned 1 exit status
Makefile:302: recipe for target '../bin/x86-release/libtrikControl-x86.so.1.0.0' failed
make[1]: *** [../bin/x86-release/libtrikControl-x86.so.1.0.0] Error 1
make[1]: Leaving directory '/home/natasha/trikRuntime/trikControl'
Makefile:166: recipe for target 'sub-trikControl-make_default' failed
make: *** [sub-trikControl-make_default] Error 2
```

dvvrd, 2016-07-06:
Try to make `git pull` and rebuild

NataliaZabelina, 2016-07-06:
With latest changes I was able to build 3.1-dev. Thank you.

closed 2016-07-06 by dvvrd