trikset/trikRuntime #313: Неверное выполнение программ передачи и приема сообщений

shirokolobov, 2019-06-27:
Ожидаемый результат: 
1010112
43

В большинстве случаев, так и есть, но иногда бывает:
101011 2
86

[image]

[передача_и_прием.zip](https://github.com/trikset/trikRuntime/files/3336323/_._.zip)

iakov, 2019-06-28:
Чтобы распаковать на линуксе: `unzip -O cp866  ~/Downloads/_._.zip `

iakov, 2019-06-28:
Переформулирую: почему-то приходит пробел лишний в каких-то случаях. Или это пустая строка при timeout ? @shirokolobov , а есть логи с контроллера? Там что-то может быть полезное в файле trik.log

shirokolobov, 2019-06-28:
Нет, но постараюсь воспроизвести.

iakov, 2019-07-02:
We already have #263 :) But we need examples, not complaints.