trikset/trikRuntime #478: Studio crashes when while True:

khodand, 2020-01-17:
Studio crashes with no output when you try to execute such code:
```python
import sys
import time
import random
import math

class Program():
  __interpretation_started_timestamp__ = time.time() * 1000
  def execMain(self):
    i = 0;
    print("start")
    while True:
      print(i)
      i += 1
    return

def main():
  program = Program()
  program.execMain()

if __name__ == '__main__':
  main()
```

anastasiia-kornilova, 2020-01-25:
Is it bug for runtime? Or it should be moved to TS issues?

iakov, 2020-01-25:
To check: if this code can be interrupted from TS when is being run on controller. @khodand

iakov, 2020-01-26:
For me works in 2D (current master): `while True: pass`

khodand, 2020-01-26:
> 
> 
> For me works in 2D (current master): `while True: pass`

It seems like this is failing while printing in a loop

khodand, 2020-01-27:
> 
> 
> To check: if this code can be interrupted from TS when is being run on controller. @khodand

No, it isn't

iakov, 2020-01-29:
Reproducible in JS, but seems to be a TS issue

PR #496 (2020-02-01): While True fix 
Fix #478
closed 2020-02-01 by iakov